douggilliland/R32V2020 #2: Confusion about jump instructions

mjgpy3, 2019-04-29:
In the opcodes document, jumping to a subroutine is stated as

Jump to   subroutine | JSR | 1 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | X | X | X | X | X | X | X | X | (rX) | X | X | X | X | X | X | X | X | X | X | X | X
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

Does `rX` indicate a register? I figured jumping might happen to labels, is that true?

douggilliland, 2019-04-29:
Yes, rX is a register (any of the ones that are valid).

Yes, jumps always have an associated label. They are non-conditionals. Branches are conditional based on the contents of the Condition Code Register (I forget which rX number that is , maybe r3?).

mjgpy3, 2019-04-30:
@douggilliland thanks, the example should clear this up! I'll probably have more questions later but I have enough to go off of at the moment.

closed 2019-04-30 by mjgpy3