douggilliland/R32V2020 #10: Dig up RAM initialization file format

mjgpy3, 2019-04-30:
@douggilliland

douggilliland, 2019-04-30:
From https://www.intel.com/content/www/us/en/programmable/support/support-resources/design-software/max_plus-ii/inramrom.html

> For ROM functions, you must specify the filename of an initial memory content file in the Intel hexadecimal format (.hex) or the Altera® Memory Initialization File (.mif) format in the Generic Map Clause, with the LPM_FILE parameter.

Intel Hex format is described here: http://www.keil.com/support/docs/1584/

The Intel HEX file is an ASCII text file with lines of text that follow the Intel HEX file format. Each line in an Intel HEX file contains one HEX record. These records are made up of hexadecimal numbers that represent machine language code and/or constant data. Intel HEX files are often used to transfer the program and data that would be stored in a ROM or EPROM. Most EPROM programmers or emulators can use Intel HEX files.

Record Format
An Intel HEX file is composed of any number of HEX records. Each record is made up of five fields that are arranged in the following format:

:llaaaatt[dd...]cc
Each group of letters corresponds to a different field, and each letter represents a single hexadecimal digit. Each field is composed of at least two hexadecimal digits-which make up a byte-as described below:

: is the colon that starts every Intel HEX record.
ll is the record-length field that represents the number of data bytes (dd) in the record.
aaaa is the address field that represents the starting address for subsequent data in the record.
tt is the field that represents the HEX record type, which may be one of the following:
00 - data record
01 - end-of-file record
02 - extended segment address record
04 - extended linear address record
05 - start linear address record (MDK-ARM only)
dd is a data field that represents one byte of data. A record may have multiple data bytes. The number of data bytes in the record must match the number specified by the ll field.
cc is the checksum field that represents the checksum of the record. The checksum is calculated by summing the values of all hexadecimal digit pairs in the record modulo 256 and taking the two's complement.

douggilliland, 2019-04-30:
Intel Hex format is probably the better choice since it would compile with other FPGA architectures.

douggilliland, 2019-04-30:
I am sure this would work too with the advantage that it would not use any BlockRAM and we'd have more space for data and stack spaces. But I could see it might eat up a lot of logic area quickly and it could only be loaded when the FPGA itself is compiled.

I think we should probably stick with the BlockRAM approach for now since a bootloader could fairly easily load the Instruction RAM before starting up the RISC CPU.

douggilliland, 2019-05-25:
HEX is working well for Instructions.
Need to check data HEX file.

closed 2019-06-02 by douggilliland