douggilliland/R32V2020 #11: Hardware purchased

douggilliland, 2019-04-30:
@mjgpy3 ORDERED FOR YOU

https://www.ebay.com/itm/Altera-Cyclone-IV-FPGA-Learning-Board-EP4CE6E22C8N-and-USB-Blaster-Programmer/254129023580?hash=item3b2b452a5c:g:p7cAAOSw--xb1ozx

douggilliland, 2019-04-30:
Video series on this card on YouTube

https://www.youtube.com/watch?v=hwnvghxc5nU&list=PLn__0BqzWEWPk0-VW3kEpMNA81sIbhj6n

douggilliland, 2019-05-01:
Got note from seller: hi ,dear customer, first of all, thanks for your new order from my store.
unfortunately , the last pc couldn't be found from our warehouse, lost by them . so, I'm afraid I have to issue the full refund and cancel this order sadly .
sincerely sorry for the trouble and inconvenience cause to you . wish you could understand .
many thanks.

douggilliland, 2019-05-01:
Will order from a different seller.

mjgpy3, 2019-05-01:
@douggilliland hahah.

douggilliland, 2019-05-03:
@mjgpy3 - Cra Ze Ape and I just bought the same FPGA development boards from Ali Express. If it's good (which it looks really good) I will get you one, too.

Check it out. Nice set of features plus a much bigger FPGA. Lot's of I/O that we can get to.

https://www.aliexpress.com/item/E22-core-bottom-board-fpga-development-board-cyclone-iv-board-EP4CE22F17C8N-altera-fpga-board-altera-board/32859908966.html?spm=2114.search0104.3.14.181b25137X4wbY&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10130_10068_10890_10547_319_10546_317_10548_10545_10696_453_10084_454_10083_10618_10307_537_536_10059_10884_10887_321_322_10103,searchweb201603_52,ppcSwitch_0&algo_expid=46ef0678-880c-4683-baa5-d767d1c93751-2&algo_pvid=46ef0678-880c-4683-baa5-d767d1c93751

douggilliland, 2019-05-03:
Features list:
* DC5V input
* FPGA: [EP4CE22F17C8N]
** Number of Logic Elements: 22320
** Total Memory: 594 kbit (1Kx9 blocks)
* SDRAM: 256M bit 
* SPI FLASH configuration storage : 64M bit
* ​JTAG: Used to download sof file to FPGA , or download jic to FLASH save program.
* 50M CLK input 
* 2 key input .key2 ,key3 
* One button to re-configure FPGA
* one LED output , use test board display.
* BANK1, BANK2, BANK3 VCCIO  can set 2.5V or 3.3V 
* BANK4, BANK5, BANK6 VCCIO  can set 2.5V or 3.3V 
* 94 I/O PINS can output or input. have 13 PINs only input CLK.
* PC PS/2 port
* VGA port
**16 bit 65536 color
* SD card 
* LCD12864/1602 LCD Display connector
* LED 8 digit, 7 segment display
* LEDs 1X8
* USB-COM port
* Buzzer
* 8X8 matrix LED 
* 3X3 key input
* 1X4 key input
* switch 1X8 input
* A/D, TLC549
* D/A, DAC7512
* IR input
* DS18B20 temperature sensor

douggilliland, 2019-05-25:
Bought two other FPGA cards. 

https://www.aliexpress.com/item/usb-blaster-EP4CE10E22C8N-altera-fpga-board-altera-board-fpga-development-board-cyclone-IV-board/32813736111.html?spm=a2g0s.9042311.0.0.19dd4c4dsYjyrR

and

https://www.aliexpress.com/item/USB-Blaster-altera-fpga-board-altera-board-fpga-development-board-EP4CE6E22C8N-board-cyclone-IV-kit/32813061054.html?spm=a2g0s.9042311.0.0.19dd4c4dsYjyrR

AliExpress is very cheap but very slow. R32V2020 will run on any of these or the other board.

First board tracks to the US. Hopefully it will be here soon.