douggilliland/R32V2020 #12: LLVM Compiler Infrastructure

douggilliland, 2019-04-30:
In the mind as target...
 
https://llvm.org

Need to crank out basic functionalty first.

douggilliland, 2019-05-24:
@mjgpy3 Good video on LLVM

https://www.youtube.com/watch?v=a5-WaD8VV38

douggilliland, 2019-05-24:
A video on the LLVM IR. 

https://www.youtube.com/watch?v=m8G_S5LwlTo

Probably the core work that would need to be done is on the IR?

douggilliland, 2019-05-24:
@mjgpy3 
Compiler Explorer looks like a fantastic tool to see what the compiler does on the fly.

douggilliland, 2019-05-24:
A helpful video with RISC-V as the example:

https://www.youtube.com/watch?v=AFaIP-dF-RA

douggilliland, 2019-06-11:
@mjgpy3 Compiler Explorer with IR output added.
[image]

douggilliland, 2019-08-11:
@mjgpy3 I have probably watched 60-80 hours of videos on LLVM and I don't feel any closer to being able to figure out how to convert LLVM IR into assembly code.

mjgpy3, 2019-08-12:
@douggilliland haha, yeah. It's always struck me as being complicated as well...

mjgpy3, 2019-08-12:
@douggilliland did you see this one? https://www.llvm.org/docs/WritingAnLLVMBackend.html

douggilliland, 2019-08-13:
@mjgpy3 
Very kewl doc! 

Thanks.