douggilliland/R32V2020 #4: Make Listfile

mjgpy3, 2019-04-30:


mjgpy3, 2019-04-30:
For example: https://stackoverflow.com/questions/16154870/how-to-read-a-nasm-assembly-program-lst-listing-file

douggilliland, 2019-04-30:
https://github.com/douggilliland/R32V2020/issues/8#issuecomment-487903181

douggilliland, 2019-05-19:
I could definitely use one of these...

douggilliland, 2019-05-26:
List file should only have the opcode hex value, not the whole HEX record line.
```
start:
00000000	:0400000000000000FC		nop
00000001	:0400010000000000FB		nop
next:
00000002	:0400020000000000FA		nop
00000003	:04000300C0FFFFFF3C		bra next		; Relative branch (-3 in this case)

```

douggilliland, 2019-05-26:
Should look like (manually edited the above):
```
start:
00000000	0000000000		nop
00000001	0000000000		nop
next:
00000002	0000000000		nop
00000003	00C0FFFFFF		bra next		; Relative branch (-3 in this case)
```

closed 2019-05-26 by mjgpy3

douggilliland, 2019-05-27:
@mjgpy3 
Very nice!

; CCR Tests
start:
00000000	00000000		nop
00000001	20800000		ads r8,r0,r0
00000002	20801000		ads r8,r0,r1
00000003	20802000		ads r8,r0,r2
00000004	22300000		cmp r0,r0
00000005	22301000		cmp r0,r1
00000006	22310000		cmp r1,r0
00000007	C0FFFFF9		bra start

commit 6abb92f1 (2019-05-26): Resolve #4: always write a list file.
commit 319131d6 (2019-05-26): Fix #4: Only show op hex rather than full intel instruction.