douggilliland/R32V2020 #18: Representing Constants in the Assembler

douggilliland, 2019-05-02:
@mjgpy3 I think you've got the Instruction part of the assembler in good shape.

How about tackling the data memory? Some portion of the data memory will contain constants.

For instance, the string "Hello, World!" needs to get stored into Data RAM as a constant. The address of the constant acts like a label does for the Instruction memory.

This ends up resulting into a separate .HEX file which gets loaded into the Data Memory at download time. Very similar to the process for instructions.

There's a lot of syntaxes out there. Here's one:

https://docs.oracle.com/cd/E26502_01/html/E28388/eoiyg.html

We probably need something for text, data (byte, short, long) storage but not much else.

douggilliland, 2019-05-02:
For simplicity, the data start address can be a directive. That way it can be put at the start of data memory, end of data memory, or anywhere.

Bytes and shorts should be long word aligned for the first element. This implies that if there are, for instance, three bytes in an array, that they are packed into 24-bits and the next location is the long address

Ex: .
.byte 0x12,0x34,0x56
.short 0x1234
.long 0x98765432,1234578

Would go at (say) 
(3) bytes are packed into a long with the value 0x12345600 to fill out a long (starting at 0)
(1) short would be packed into a long with the value 0x1234000 (starting at 1 - since all data is longs)
Of course the long would be located at 2.

The endianswap instruction can take care of byte ordering for us since we'd probably want to deal with the 0x12 before the 0x34...

douggilliland, 2019-05-02:
This does "waste" some space but having everything packed into longs makes a lot of sense. Not dealing with byte, word, long access to memory is a good simplification as well.

douggilliland, 2019-05-02:
.string "Hello, World!" 
would pack into 
0000 "Hell"
0001 "o, W"
0002 "orld"
0003 "!\0\0\0"
\0 is null ;)

douggilliland, 2019-05-02:
Variables don't need data packed into them but they do need addresses assigned.

mjgpy3, 2019-05-02:
@douggilliland are there special instructions to operate on the constants? What would using the `"Hello, World!"` constant look like in actual code?

Also, there's not intersection in address space, right? In other words, the first constant in this hex file is address 0 in "constant space" and the first instruction in the assembly is address 0 in "code space"?

douggilliland, 2019-05-02:
@mjgpy3 Yep. We are non-Von Neumann 

https://web.archive.org/web/20131212205159/http://aws.linnbenton.edu/cs271c/markgrj/

douggilliland, 2019-05-02:
Instruction Space is read only and completely separate from Data Space.

douggilliland, 2019-05-02:
IE, no self-modifying-code is possible.

mjgpy3, 2019-05-02:
@douggilliland gotcha, thanks. What does accessing the constants from code look like?

douggilliland, 2019-05-02:
Constants are very similar to addresses for jumps.

The code would need to know the address of the constant and be able to do a read from that location with a move.

I can write some psuedo-code later (after work).

mjgpy3, 2019-05-03:
I found an example on page 2 of this [document][1]

  [1]: https://user.eng.umd.edu/~blj/RiSC/RiSC-ex.pdf

douggilliland, 2019-05-03:
It really is just a label which really is just an address.

Our CPU can only get half of the 32-bit address at a time and it can only do that with load immediate of each half at a time into a register.

The immediate value is known because the address is known.

mjgpy3, 2019-05-04:
@douggilliland how shall we represent string characters in the data hex file?

Should we use a byte representing the ASCII character or some other encoding?

Also, does the null character basically just encode to the null byte?

Thanks.

douggilliland, 2019-05-04:
Two hex digits = ASCII character. 0x31 = ASCII for the number 1.
Yes, null is 0x00.

mjgpy3, 2019-05-04:
@douggilliland thanks!

mjgpy3, 2019-05-11:
@douggilliland what is the representation of an empty string? Is it just `0000`?

PR #26 (2019-05-11): Basic constant parsing
Resolved #18
closed 2019-05-11 by mjgpy3