douggilliland/R32V2020 #23: Use hash to denote immediate values

mjgpy3, 2019-05-04:


douggilliland, 2019-05-23:
@mjgpy3 Is this in the code? If so, what is the format?

mjgpy3, 2019-05-24:
@douggilliland right now, loading intermediates just use syntax like `0xab`. You had mentioned that you wanted them to be denoted with a `#` character like `#0xab`. This issue was a reminder to implement that.

douggilliland, 2019-05-25:
**VHDL Changes needed for Immediate loads**

- Change registers r8-r15 from REG32 to 2 of REG16
- Immediate loads are not allowed on r0-r2 - These are constants
- Immediate loads are not allowed on r3 (CCR only changes by ALU operations).
- Immediate loads could be allowed on r4-r6 but they could result in a non-atomic operation (better to load them from r8-r15)
- Immediate loads are never allowed on r7 (changing half the PC would be bad)
- Separate write controls for each half 
- Muxes to shift up 16-bit for liu (Load Immediate Upper short)

douggilliland, 2019-05-25:
Decided that immediate instructions destination registers need to be limited to r8-r15 only.
LIL and LIU are the only two instructions which load the register file from the Instruction ROM data values.

Still need to add muxes.

douggilliland, 2019-05-25:
Immediates are now working (in the FPGA).

douggilliland, 2019-05-25:
[image]

closed 2019-06-02 by mjgpy3