douggilliland/R32V2020 #28: Flow Control (BRA, JMP, JSR, etc)

douggilliland, 2019-05-13:
Thinking about how to best do flow control. Many options. All are viable. Some make code more complicated. Some make hardware more complicated. All should respect the design constraint that all opcodes are 32-bits. 

- JMP should jump to a 32-bit address. That address has to be in a register. JMP then consists of loading the address into a register and then moving the address from the register into the Program Counter register.
- JSR could push Program Counter onto Stack and then jump to address (rX).
- BRA could be set up to be relative branching with bottom 16 bits of the opcode being the relative offset. Would need to handle both positive and negative numbers (sign extend d15 into d16-d31). Would need an adder added (could it use ALU?).
- BCx - Could also be relative or absolute. 

**Tradeoff of relative vs absolute**
Absolute address would have to be loaded into one of the registers. Loads are half of the register at a time. This means that whether or not the branch is taken it would need to be loaded by prior instructions into a register.

douggilliland, 2019-05-17:
@mjgpy3  - Added JMP instruction to opcodes. Different from JSR since it doesn't save the PC to the stack. Here is the simplest piece of code possible:
```
start: ; label is not actually used by the code
  jmp r0    ; r0 always has zero which is the start of the instruction memory
```

More on new JMP instruction:
- JMP should jump to a 32-bit address. That address has to be in a register.  JMP consists of moving the address from the register into the Program Counter register. The effect is that the next address will be address 0. C language equivalent is:
`while(1);`
If the assembler sees a label on a jump it could know that it has to load the label into a register but it wouldn't know which register (r8-r15) that it is allowed to use so this isn't something that the assembler could do. This tells me that the register would have to be assigned by the person writing the assembly code. This is cumbersome for the moment but something that a compiler would do since it controls the registers.

douggilliland, 2019-05-17:
Flow control instructions are by far the hardest part of this architecture. They break any pipeline, hence people shove in NOPs for as many levels as pipelines are deep.

In our initial case we have a pipeline "sorta". Not really a pipeline because an instruction goes through all of the stages before the next instruction is done. A real pipeline (and we will get there eventually) will allow for the stages to be loaded with each clock.

FPGA simulations show the ALU runs at 50 MHz. We have these stages in the instruction path.

https://raw.githubusercontent.com/douggilliland/R32V2020/master/Architecture/Architecture%20Specs/Instruction_Pipeline.png

I think I count about 6 stages. If we move the instruction through one stage at a time and we can run at 50 MHz that's around 8 MHz instruction execution time.

douggilliland, 2019-05-17:
@mjgpy3 Interestingly, with the architecture we have we really don't need a JMP. All I have to do is add r0 to the address I want to go to and store it in the PC. 

Going to use that as the first code example.

```
start:
  ads r7,r0,r0  ; jmp to 0x00000000 (start)
```

douggilliland, 2019-05-17:
@mjgpy3 Success in compiling first program.

douggilliland, 2019-05-17:
@mjgpy3 The hex file output needs more stuff. Example: 

This example shows a file that has four data records followed by an end-of-file record:

:10010000214601360121470136007EFE09D2190140
:100110002146017E17C20001FF5F16002148011928
:10012000194E79234623965778239EDA3F01B2CAA7
:100130003F0156702B5E712B722B732146013421C7
:00000001FF

See: https://en.wikipedia.org/wiki/Intel_HEX

douggilliland, 2019-05-17:
@mjgpy3 - Here's the actual Intel Hex spec.

https://web.archive.org/web/20160607224738/http://microsym.com/editor/assets/intelhex.pdf

We want the 32-bit version.

douggilliland, 2019-05-17:
@mjgpy3 
Found a hex editor that can read in the assembly hex and make an Intel hex file.

https://mh-nexus.de/en/downloads.php?product=HxD20

Using that for the moment. Produced output:

:020000040000FA
:0A00000020700000300000000D0A1F
:00000001FF

Seems probably correct. Will try with FPGA tool and see if it reads it in.

douggilliland, 2019-05-18:
@mjgpy3 
Got first program running (apparently).
Routed the upper 8-bits of the opcode to the first 2 of seven segment LEDs and the bottom 8 address bits to the second 2 of seven segment LEDs.
Display shows x20 00.
x20 is the opcode.
x00 is the address of the first opcode.

Struggled with the hex file that I made. It didn't work exactly right. Found that the chip memory can be directly accessed from a Tool:
https://www.intel.com/content/www/us/en/programmable/quartushelp/18.1/index.htm#reference/shortcut/gl_short_list_keyboard_d6228e1182.htm
Poked 20 70 00 00 to the first long and then 00 to the rest of the longs.

My hex file had the CR and LF in the file and it was offset from the start by two bytes.

Can you help clean all of this up? I want to do this with the assembler...

Next going to try NOP followed by jump to start.

douggilliland, 2019-05-18:
I was able to dump my program from the memory and this is what I got for the first line:
:04000000207000006C
':' = record mark
’04’ = Extended Linear Address Record
'00' = record length - not sure why this is zero (seems like it should be 1
'0000' = address bits 
'0000' = 16 address bits
'20700000' = ads r7,r0,r0 opcode

douggilliland, 2019-05-25:
Flow control code is created and partly tested.

**Currently**

- Writing to r7 will do flow control. Three easy moves (implemented as adds) will change PC. This doesn't require a jump or branch instruction

**Examples**
ADS r7,r0,r0 ; jumps to address 0
ADS r7,r1,r0 ; jumps to address 1
ADS r7,r1,r1 ; jumps to address 2

**Need to do**

- Get immediates working so that I can put in an address other than the "easy" ones.

douggilliland, 2019-05-26:
@mjgpy3 Question about the assembler and address calculations.

I can only load a half of an address via immediate loads (LIU, LIL) at a time. Does the assembler let me do that?

Example:

start:
	liu r8,start	; Should set the upper 16-bits of r8 to the upper 16 bits of start
	lil r8,start		; Should set the lower 16-bits of r8 to the lower 16 bits of start
	jmp r8		; Copy r8 into r7 - causing a jump to start

What do you think?

douggilliland, 2019-05-26:
I get this error when I try to compile the code.
```
start is not a valid immediate value

Line 1:
        liu r8,start    ; Should set the upper 16-bits of r8 to the upper 16 bits of start
```

douggilliland, 2019-05-26:
@mjgpy3 

I am re-thinking this whole subject completely to be closer to what you already have in the assembler. I want low impact to the assembler at this point.

Basic issue is that the address is 32-bits so jumps (as they were originally designed) requires:

- Loading upper and lower halves via LIU, LIH (as above) - 2 instructions
- Followed by the jump - a 3rd instruction
- Would consume a GP register - no good justification for this concession, IMO

Fixing this by a tradeoff:

- Delete JMP, JSR, RTS opcodes (may add them back in later on)
- Do all flow control as Branch opcodes
- All Branch opcodes are relative to the program counter
- Make Branch instructions all 24-bit relative to the current PC
- BRA 0x1 would go to the next line
- Hardware will sign extend branch offset up to 32-bits to allow branching backwards
- Limit: Only can jump within 8M instructions (+/- 23 bits). Not a serious limitation in this or any for that matter FPGA design. If we write a program with 8M instruction it will then have to span multiple assembly files and need a linker.
- Longer jumps can be accommodated via LIU, LIH and then moving the register to r7 (via ADS r7,rX,r0)
- Longer jumps would be rare but could happen with programs stored in external SDRAM

douggilliland, 2019-05-26:
Opcode changes are

- Addition of new Bxx instructions.
- 24-bit relative address (positive and negative) 

This requires changes to the assembler. 

- Use 24 bits relative - calculate difference between current address and the label
- Positive and negative offsets (instead of absolute addresses)

douggilliland, 2019-05-26:
@mjgpy3 
C008 works!!!
[image]

Still need to test other condition codes. Closing this one for now since this is an entirely new piece of logic working.

closed 2019-05-26 by douggilliland

douggilliland, 2019-05-26:
@mjgpy3 This code works.
```
; Code that jumps all over the place
start:
00000000	:0400000000000000FC		nop
00000001	:0400010000000000FB		nop
next:
00000002	:0400020000000000FA		nop
00000003	:04000300C000000336		bra note2
note1:
00000004	:0400040000000000F8		nop
00000005	:0400050000000000F7		nop
note2:
00000006	:0400060000000000F6		nop
00000007	:0400070000000000F5		nop
00000008	:04000800C0FFFFFC3A		bra note1
```
[image]

mjgpy3, 2019-05-26:
@douggilliland that's awesome

douggilliland, 2019-05-27:
@mjgpy3 
If you are at all curious about how to make a machine Turing complete and how Conditionals work here's a good video. His whole series is excellent. He makes a 4-bit computer on breadboards with tons of LEDs to see how it work.

https://www.youtube.com/watch?v=Zg1NdPKoosU

douggilliland, 2019-05-27:
Here is his playlist of making a 4-bit computer on a breadboard.

https://www.youtube.com/playlist?list=PLowKtXNTBypGqImE405J2565dvjafglHU

douggilliland, 2019-05-27:
Here's the specific video on Turing Complete subject:

https://www.youtube.com/watch?v=AqNDk_UJW4k

douggilliland, 2019-05-27:
@mjgpy3 
Wow, that was incredibly easy. We are now Turing Complete.
[image]

douggilliland, 2019-05-27:
All condition codes work (at least with ADS and CMP opcodes).

[image]

douggilliland, 2019-05-27:
@mjgpy3 
Checked flow control branching (above just checked the flags).
Added BNE (Branch Not Equal) test
Code is:
```
; CMP and Bxx Checks
start:
	cmp r0,r0
	bne flagFail
	beq	skip1
	bra	flagFail
skip1:
	cmp r0,r1
	beq flagFail
	bne	skip2
	bra	flagFail
skip2:
	bra start
flagFail:
	hcf
```
Works!
[image]