douggilliland/R32V2020 #33: Create opcode constants table

douggilliland, 2019-05-24:
@mjgpy3 
I have a VHDL pattern for the opcodes. It defines constants for the various opcodes. 
Looks like:
constant NOP	: std_Logic_Vector(7 downto 0) := "00000000";
constant HCF	: std_Logic_Vector(7 downto 0) := "00000001";
constant RES	: std_Logic_Vector(7 downto 0) := "00000010";
-- ALU
constant ADS	: std_Logic_Vector(7 downto 0) := "00100000";
...

Would be cool if the whole table dropped out of the assembler. Would make it easier to keep things in sync.

douggilliland, 2019-05-24:
Here's the current list I manually made (and update manually):
-- System
constant NOP	: std_Logic_Vector(7 downto 0) := "00000000";
constant HCF	: std_Logic_Vector(7 downto 0) := "00000001";
constant RES	: std_Logic_Vector(7 downto 0) := "00000010";
-- ALU
constant ADS	: std_Logic_Vector(7 downto 0) := "00100000";
constant MUL	: std_Logic_Vector(7 downto 0) := "00100001";
constant CMP 	: std_Logic_Vector(7 downto 0) := "00100010";
constant ORS 	: std_Logic_Vector(7 downto 0) := "00101000";
constant ARS 	: std_Logic_Vector(7 downto 0) := "00101001";
constant XRS 	: std_Logic_Vector(7 downto 0) := "00101010";
constant LS1 	: std_Logic_Vector(7 downto 0) := "00110000";
constant RS1 	: std_Logic_Vector(7 downto 0) := "00110001";
constant LR1 	: std_Logic_Vector(7 downto 0) := "00110010";
constant RR1 	: std_Logic_Vector(7 downto 0) := "00110011";
constant RA1 	: std_Logic_Vector(7 downto 0) := "00110100";
constant ENS 	: std_Logic_Vector(7 downto 0) := "00111000";
-- Immediate Opcodes
constant LIL 	: std_Logic_Vector(7 downto 0) := "01000000";
constant LIU 	: std_Logic_Vector(7 downto 0) := "01000001";
-- Load/Store Data Memory Opcodes
constant LDB 	: std_Logic_Vector(7 downto 0) := "01100000";
constant SDB 	: std_Logic_Vector(7 downto 0) := "01100001";
constant LDS 	: std_Logic_Vector(7 downto 0) := "01100010";
constant SDS 	: std_Logic_Vector(7 downto 0) := "01100011";
constant LDL 	: std_Logic_Vector(7 downto 0) := "01100100";
constant SDL 	: std_Logic_Vector(7 downto 0) := "01100101";
-- Peripheral I/O Opcodes
constant LPB 	: std_Logic_Vector(7 downto 0) := "10000000";
constant SPB 	: std_Logic_Vector(7 downto 0) := "10000001";
constant LPS 	: std_Logic_Vector(7 downto 0) := "10000010";
constant SPS 	: std_Logic_Vector(7 downto 0) := "10000011";
constant LPL 	: std_Logic_Vector(7 downto 0) := "10000100";
constant SPL 	: std_Logic_Vector(7 downto 0) := "10000101";
-- Stack Opcodes
constant PSS 	: std_Logic_Vector(7 downto 0) := "10100000";
constant PUS 	: std_Logic_Vector(7 downto 0) := "10100001";
constant SSS 	: std_Logic_Vector(7 downto 0) := "10100010";
constant LSS 	: std_Logic_Vector(7 downto 0) := "10100011";
-- Flow Control
constant JSR 	: std_Logic_Vector(7 downto 0) := "11000000";
constant RTS 	: std_Logic_Vector(7 downto 0) := "11000001";
constant BRA 	: std_Logic_Vector(7 downto 0) := "11010000";
constant BCS 	: std_Logic_Vector(7 downto 0) := "11010001";
constant BCC 	: std_Logic_Vector(7 downto 0) := "11010010";
constant BEZ 	: std_Logic_Vector(7 downto 0) := "11010011";
constant BE1  	: std_Logic_Vector(7 downto 0) := "11010100";
constant BOV 	: std_Logic_Vector(7 downto 0) := "11010101";
constant BEQ  	: std_Logic_Vector(7 downto 0) := "11010110";

mjgpy3, 2019-05-24:
@douggilliland sounds like a great idea!

mjgpy3, 2019-05-25:
@douggilliland I took care of this. I'll update the wiki page to show how to run.

closed 2019-05-25 by mjgpy3

douggilliland, 2019-05-25:
Awesome! Very nice to have that ability. 

I am trying to figure out how to make the constant values into globals for the VHDL code. I think they have some sort of "package manager" that allows that. For the moment I will copy/paste it where it needs to go.

Thanks!

commit abd9d1b7 (2019-05-25): Resolve #33. Generate constants file when `--gen-constants` flag is passed.