douggilliland/R32V2020 #45: Tried making R7_DEST

douggilliland, 2019-06-06:
C:\Users\HPz420\Documents\GitHub\R32V2020\VHDL\Programs\C040-BSR_RTS_Test>python ..\..\..\Assembler\assembler.py C040-BSR_RTS_Test.asm
Unexpectedly failed to parse line

Line 7:
        rts     r7,SAR,r0       ; need no param version

douggilliland, 2019-06-06:
C:\Users\HPz420\Documents\GitHub\R32V2020\VHDL\Programs\C040-BSR_RTS_Test>python ..\..\..\Assembler\assembler.py C040-BSR_RTS_Test.asm
Unexpectedly failed to parse line

Line 7:
        rts     ; need no param version

douggilliland, 2019-06-06:
ReTurn from Subroutine (RTS) should set the destination register to r7.

douggilliland, 2019-06-06:
@mjgpy3 

I created a R7_DEST but apparently don't know how to make it hooked up.

douggilliland, 2019-06-07:
@mjgpy3 
I did some stuff to the assembler and checked it in.

I got it kinda working but didn't know how to get r7 as the destination. I just coded it like the NOP. 

Can you fix it to use r7 as the destination?

douggilliland, 2019-06-07:
@mjgpy3 

I got BSR working great. I temporarily set aside RTS until you can help me but it turns out that doing:

  pus r7

does the same thing as RTS. Maybe there's a clue there on implementation? We could make RTS a psuedo-operation. Use the same coding as pus r7 but make RTS a shortcut which automatically uses r7? What do you think?

Take a look at C043. It's working well.

mjgpy3, 2019-06-07:
@douggilliland I'm on it.

mjgpy3, 2019-06-07:
@douggilliland the new form `BIN_CONST` requires that the data at bits 23, 19 and 15 are valid registers in `ops.csv`

https://github.com/douggilliland/R32V2020/blob/master/Assembler/assembler.py#L118-L120

closed 2019-06-07 by mjgpy3

mjgpy3, 2019-06-08:
(I just wanted to mention that since it's the first form that looks beyond the category and opcode bits)

commit 811382ca (2019-06-07): Resolve #45: make RTS work.

Adds a new special forms for binary operations where every register is
a constant (i.e. the instruction appears as a no-arg but actually has 3
static arguments).

@douggilliland your attempt didn't work because this doesn't fit the
form of anything thus far.