douggilliland/R32V2020 #46: Immediate Value length is 20 bit

douggilliland, 2019-06-09:
@mjgpy3 
I was wrong when we did the immediate value stuff.

The immediate value is 20 bits not 24 bits.

Assembler needs to be fixed.

douggilliland, 2019-06-09:
I tried to change the assembler but I couldn't find the line where we set the immediate value.

douggilliland, 2019-08-04:
I fixed this one. Source:

; Test 20 bits immediate load values

start:
	lix		r8,1000000

Result:

; Test 20 bits immediate load values

start:
00000000	428F4240		lix		r8,1000000

mjgpy3, 2019-08-10:
LIL length test should still be failing. (it should be 16 bits)

closed 2019-08-10 by mjgpy3

douggilliland, 2019-08-10:
@mjgpy3 
This now works (in hardware)!
start:
	lix		r8,0x1234
	addi		r8,r8,1
	subi		r8,r8,4
	bsr		wr7Seg8Dig

douggilliland, 2019-08-11:
@mjgpy3    Here is the value of what we did yesterday.
 All of the commented code gets replaced with one line ---
;	lix		r9,0x57
;	xor		r9,r9,MINUS1
;	add		r9,r9,ONE
;	add		r8,r8,r9
;	lix		r9,0x0F
;	and		r8,r8,r9
	subi	r8,r8,0x57

Very kewl. Much more tight and frees a general purpose register.

mjgpy3, 2019-08-11:
Nice!

commit c0cc082d (2019-08-10): Resolve #46: Ensure new immediate ops have correct size.