douggilliland/R32V2020 #48: Data RAM access changes

douggilliland, 2019-07-27:
@mjgpy3 I changed the Data RAM accesses to be byte/word/long addresses.

It should fill the exact same way from the assembler but the labels are off by power of 4 in their location/value.

For instance:
firstVal: .long 0x12345678
secVal: .long 0xDEADBABA

secVal should be at address 0x4 (worked before but not after my hardware change) but it is 0x1.

00000020	42800000		lix	r8,firstVal.lower
00000021	42800001		lix	r8,secVal.lower

Can you shift the data label values left by 2?

The reason I did this was that it was too painful to pack/unpack bytes into longs. For instance, readLine got really painful when I wanted to implement backspaces. It had to backtrack differently for longs that were already stored. If it's bytes then it "just works".

mjgpy3, 2019-07-28:
@douggilliland does this look right for hello world?
``` hex
:0400000048656C6C77
:040004006F2C2057E6
:040008006F726C6443
:04000C0000000000F0
```

douggilliland, 2019-07-28:
@mjgpy3 I can't remember where hello world is ;)

douggilliland, 2019-07-28:
@mjgpy3 I check in C073 which shows the problem.

douggilliland, 2019-07-28:
It still doesn't work. 

firstLong:	.long 0x00000001
secondLong:	.long 0x00000002

start:
00000000	42800000		lix		r8,firstLong.lower
00000001	**42800001**		lix		r8,secondLong.lower
	
firstLong:	.long 0x00000001
secondLong:	.long 0x00000002

start:
00000000	42800000		lix		r8,firstLong.lower
00000001	**42800004**		lix		r8,secondLong.lower

douggilliland, 2019-07-28:
Something about one of the changes broke other stuff.

I only get part of the prompt now.

douggilliland, 2019-07-28:
@mjgpy3 Weird since I see it in the data file so it has to be something I broke.

douggilliland, 2019-07-28:
@mjgpy3 
[image]

douggilliland, 2019-07-28:
@mjgpy3 
The data should be contiguous.

I think when you did one of the left shifts it should not have been done.
The one for the address of the label was correct to do (the last one you did).
But a previous one wasn't.

douggilliland, 2019-07-28:
@mjgpy3 

I think line 428 was the commit that broke it.

[image]

douggilliland, 2019-07-28:
I backed out the change to line 428 and it works. Committed.

commit 81170263 (2019-07-28): Resolve #48 shift data addresses by 2 more bits.
commit 5314bbef (2019-07-28): Resolve #48: fix data addressing too.
closed 2019-07-28 by douggilliland