douggilliland/R32V2020 #50: Add new instruction cmpi

douggilliland, 2019-07-30:
@mjgpy3 
Can you add a new instruction "type"?

I am finding that I am doing a lot of this sequence:

    lxi r8,0x1234
    cmp r8,r9
    bxx  label

I'd like to cut this way down by having a compare register to immediate value. It would be like this:

    cmp r8,0x1234
    bxx  label

I tried to put it into the ops.csv but there are no other FORMs that match this format.

The destination register is r3 just like the other compare. But I think that may be implicit anyway. 

Also, the rSrc2 would encode the comparison register.

Also, the immediate value would be shorter than the other immediates since there's only 16-bits available instead of the 20 bits that the other immediates have.

I'd ultimately like to have immediate versions of a lot of the other operations like ADD.

mjgpy3, 2019-08-10:
Let's do `addi` next!

``` asm
addi r9,r8,0x1234
```

mjgpy3, 2019-08-10:
Some more immediate instructions to do!

douggilliland, 2019-08-10:
'''addi r8,r9,01234
cmpi r8,0x1234'''

douggilliland, 2019-08-10:
@mjgpy3 
This works!!!

	lix		r8,0x1234
	addi		r8,r8,1
	subi		r8,r8,4
	cmpi	r8,0x1231
	beq		goodVal
	lix		r8,0xBAD
	bsr		wr7Seg8Dig
	bra		forever
goodVal:
	; got to here just like it should

closed 2019-08-10 by douggilliland