douggilliland/R32V2020 #58: #define would be really helpful

douggilliland, 2019-09-06:
@mjgpy3 Something like #define could be very helpful for keeping tracks of constants - like the ports in the memory map.

Would be something like the C version:

#define _SERIAL_PORT_BASE 0x1000

Then it could be used with something like:

lix r9,_SERIAL_PORT_BASE.lower

mjgpy3, 2019-10-10:
@douggilliland would `#DEFINE` work like a standard c-style precompiler directive?

E.g. `#define SOME_A SOME_B` basically means replace every occurrence of `SOME_A` with `SOME_B` in the file? Should it span files or only work in the file containing the `#define`?

douggilliland, 2019-10-10:
@mjgpy3 
Yep, just like C. 
Doesn't probably matter if it is local to the file or persists into the include files.

mjgpy3, 2019-10-13:
Is it okay to require that `#define` comes before usage of the term it defines? I shouldn't need another pass if this is acceptable.

mjgpy3, 2019-10-13:
Also, I'm going to restrict the definable terms to alphanumeric plus `_` if that's okay.

mjgpy3, 2019-10-13:
I know C let's one do some pretty crazy stuff with these :)

mjgpy3, 2019-10-13:
Also I'm going to make this replace at the _token_ level. In other words crazy things like the following aren't allowed.

``` asm
#DEFINE dr di
addr r10, r11, 42
```

To try to get
``` asm
addi r10, r11, 42
```

commit d978c6ed (2019-10-13): Resolves #58: Implement the `#define` precompiler directive.
commit 644496fc (2019-11-11): Merge pull request #64 from douggilliland/pound_define

Resolves #58: Implement the `#define` precompiler directive.
closed 2019-11-11 by mjgpy3