douggilliland/R32V2020 #59: Assembler Python traceback

douggilliland, 2019-09-28:
@mjgpy3 
Not sure yet what line in my code caused this...

```
C:\Users\HPz420\Documents\GitHub\R32V2020\Programs\Games\G002-Tetris>..\..\..\Assembler\assembler.py G002-Tetris.asm
Traceback (most recent call last):
  File "C:\Users\HPz420\Documents\GitHub\R32V2020\Assembler\assembler.py", line 734, in <module>
    for data in constant.resolveHex():
  File "C:\Users\HPz420\Documents\GitHub\R32V2020\Assembler\assembler.py", line 287, in resolveHex
    return [int(self.longToken, 16)]
ValueError: invalid literal for int() with base 16: ''
```

mjgpy3, 2019-09-28:
@douggilliland could you share the tetris code?

douggilliland, 2019-09-28:
@mjgpy3 
Just pushed in Games folder

douggilliland, 2019-09-28:
@mjgpy3 
It usually makes a list file and when it does I can often find the error there. This one didn't update the list file at all. Weird.

douggilliland, 2019-09-28:
Maybe it died here?

```
setScreenLocation_XGA:
	push	r8
	push	r9
	lix		DAR,screenX.lower
```

douggilliland, 2019-09-28:
@mjgpy3 
Found workaround.
It likes:
`screenX:	.LONG	0x0`
It hates:
`screenX:	.LONG	0`

mjgpy3, 2019-09-29:
@douggilliland should shorts and bytes be allowed as normal integers too?

mjgpy3, 2019-10-10:
@douggilliland I'm going to close this for now as I believe I've fixed it for longs. If we want it fixed for shorts and bytes let's open another issue :)

closed 2019-10-10 by mjgpy3