douggilliland/R32V2020 #65: "Upgrade" assembler to Python 3

mjgpy3, 2019-10-13:
I hate to say it but the python folks are moving on :(

https://pythonclock.org

douggilliland, 2020-03-08:
@mjgpy3 
I upgraded to Python 3 but broke something.

Checksum should be two digits. Looks like it's now 1 digit.

start:
00000000	D4000105	bsr		clearScreen_mmXGA
; Print the title
00000001	42800000	lix		r8,0x0000			; Upper left corner of screen
``
Is making code:

```
:04000000D40001052
:04000100428000003
```
Looks like only 1 digit of checksum. Not sure what I broke in making code run under Py3.

I did replace xranges with ranges if that helps.

mjgpy3, 2020-03-18:
@douggilliland please double check that the python 3 stuff works. All my tests passed.

closed 2020-03-18 by mjgpy3

douggilliland, 2020-03-19:
@mjgpy3 Compiled and ran on the hardware!!! Thanks!!!

commit 9cf22eac (2020-03-18): Resolve #65. "Upgrade" code to python3.